chegewara/esp32-usb-v2 #2: Have option to make MSC device non-writable

kwirk, 2022-12-23:
For my use case, I wanted the MSC device to be read-only to this host accessing via USB interface. It would be useful to have some way to enable this.

I'm fairly new to using Arduino, etc. so I'm not sure of best approach (maybe I could enable within my project?), but for now I've implemented via the following hard coded change: ddfeaf1

I've tested the above commit using ESP32-S3, with SD-Card MSC interface, and Linux OS reports device is write protected, and mounts FAT32 filesystem as read-only.

chegewara, 2022-12-23:
I dont remember right now, but there is some "trick" described in tinyusb how to do it.

From my point of view this code you posted is looking good, but you can add it into your "main.ino" and will do the same (no need to change library API).

kwirk, 2022-12-24:
Thanks @chegewara, including the code from ddfeaf1 in my Arduino main project file worked.

closed 2022-12-24 by kwirk