chegewara/esp32-usb-v2 #3: Arduino: no headers files (.h) found

padaszewski, 2023-02-12:
Hi,

I'm trying to prototype with Arduino, but I can't compile the sample sketches because of the error:
`no headers files (.h) found`. In fact there are no header files.

Here is what I have done:
1. Repo downloaded,
2. Copied to Arduino/libraries
3. New project with example code (cdc)
4. Result: `usb_cdc.hpp: No such file or directory`

Any help with this?

chegewara, 2023-02-12:
Yes, sorry for that. Ive been trying to make library compatible with arduino and esp-idf, but its a big mess right now.
Arduino is using own clone of tinyusb and own config flags, so i have to rest a bit from this library and play a bit with something else.

padaszewski, 2023-02-13:
Thx for the fast answer. About half a year ago I was trying to create a prototype host device (ESP32 S3) for a custom USB device. I tried many libraries, but the only one that helped me in any way at that time was yours. It was the only one that received any messages from the device. I am not an expert in hardware or C programming. I started prototyping in Arduino, but somehow without success. Would you be able to advise if it is worthwhile to take an interest in ESP-IDF in such a usecase? And if I have a custom USB device, which CDC should I start with?
And did I understand correctly? For now you stop the development of this library?

chegewara, 2023-02-13:
Hi,
yes, this email is valid.
Sorry, but i have full time job, esp32 related, and after over 5 years with esp32 i feel a bit burned out. espressif is less and less responsive on github issues reported not mentioning forum. 
Its time to learn new SoCs and new stuff, especially i have no profits from github libraries.

djzrs, 2023-04-10:
Arduino:1.8.19 (Windows 10), 开发板："ESP32S2 Dev Module, Enabled, Disabled, Disabled, Internal USB, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None, Disabled"

keyboard:3:10: fatal error: usb_keyboard.hpp: No such file or directory

compilation terminated.

exit status 1

usb_keyboard.hpp: No such file or directory

在文件 -> 首选项开启
“编译过程中显示详细输出”选项
这份报告会包含更多信息。

chiffacff, 2023-06-30:
Hello. Do you find solutions?

lathoub, 2023-11-22:
I believe this relates to a missing `library.properties` file, that specifies a default include file (`includes=`).

See [here](https://arduino.github.io/arduino-cli/0.19/library-specification/)

It states:
`If the includes property is missing, all the header files (.h) on the root source folder are included.`

Buy there are only `.hpp` files, no `.h` files, hence the error.

I got past the missing file error (I added a usb_message.cpp and usb_message.cpp file, the .h file includes the .hpp) during compilation, only to get error stated in #8