chegewara/esp32-usb-v2 #4: Build fails with Arduinoand esp32s3 on Ubuntu Device

ZanzyTHEbar, 2023-03-15:
Hello, 

Fantastic. I normally program on Window, and have not had an issue yet with your library. 

However, i recently setup `CI` for a  new project/client - using this lib for the `HID` application and i ran into an issue. 

```
 In file included from .pio/libdeps/esp32s3_fancy_display/esp32-usb-v2/src/usb_keyboard.hpp:3,
                   from lib/FancyDisplay/src/io/usb/local_keyboard.hpp:5,
                   from src/main.cpp:6:
  .pio/libdeps/esp32s3_fancy_display/esp32-usb-v2/src/usb_hid.hpp:86:102: error: 'void tud_hid_report_complete_cb(uint8_t, const uint8_t*, uint8_t)' should have been declared inside '::'
           friend void ::tud_hid_report_complete_cb(uint8_t instance, uint8_t const* report, uint8_t len);
                                                                                                        ^
  *** [.pio/build/esp32s3_fancy_display/src/main.cpp.o] Error 1
  ========================= [FAILED] Took 54.05 seconds =========================
```

This error only occurs within the `github action` runner. On my home machine (windows  10) the project builds and runs perfectly fine.

chegewara, 2023-03-15:
Hi,
to be honest, i have no idea why you see this error (im not advanced C++ programmer) and i know nothing about github actions.
It is also possible this function definition changed or you are missing tinyusb library in your github actions.

joelsernamoreno, 2023-03-20:
Hi

I have the same error with ESP32-S2. Did you fix the error? @ZanzyTHEbar

ZanzyTHEbar, 2023-04-07:
So,, i recently installed windows 11 - not this _exact_ same issue happens on my windows 11 machine. So i can confirm without a doubt that the issue is not OS related. It's somehow related to the compiler. 

I don't have my window10 install where this worked anymore, so no idea what compiler i had.

ZanzyTHEbar, 2023-04-07:
Update-  i was able to get it to compile by commenting out
```cpp
friend void ::tud_hid_set_report_cb(uint8_t instance, uint8_t report_id, hid_report_type_t report_type, uint8_t const *buffer, uint16_t bufsize);
friend void ::tud_hid_report_complete_cb(uint8_t instance, uint8_t const* report, uint8_t len);
```
And, then making these two function public:
```cpp
virtual void _onSetReport(uint8_t report_id,
                              uint8_t const* buffer,
                              uint16_t bufsize) {}  // TODO
virtual void _onSendComplete(uint8_t const* buffer, uint16_t bufsize) {}  // TODO
```