chegewara/esp32-usb-v2 #11: Importing your repository as an Arduino Library

wbadry, 2024-03-05:
Hello,
I am quite fascinated by your library for mass storage devices based on SDMMC.

The only issue I have is that I have no clue how to make it look like a library for Arduino and try the example.

Thank you so much. You certainly invested too much time to build it that neatly.

wbadry, 2024-03-05:
I added `library.properties` missing file.

When compiling `sdcard.ino`

```bash
In file included from c:\users\wbadry\documents\arduino\libraries\esp32-usb-v2\src\usb_keyboard.hpp:3,
                 from c:\Users\wbadry\Documents\Arduino\libraries\esp32-usb-v2\src\hid\keyboard.cpp:1:
c:\users\wbadry\documents\arduino\libraries\esp32-usb-v2\src\usb_hid.hpp:86:102: error: 'void tud_hid_report_complete_cb(uint8_t, const uint8_t*, uint8_t)' should have been declared inside '::'
         friend void ::tud_hid_report_complete_cb(uint8_t instance, uint8_t const* report, uint8_t len);
                                                                                                      ^
In file included from c:\users\wbadry\documents\arduino\libraries\esp32-usb-v2\src\usb_mouse.hpp:3,
                 from c:\Users\wbadry\Documents\Arduino\libraries\esp32-usb-v2\src\hid\mouse.cpp:1:
c:\users\wbadry\documents\arduino\libraries\esp32-usb-v2\src\usb_hid.hpp:86:102: error: 'void tud_hid_report_complete_cb(uint8_t, const uint8_t*, uint8_t)' should have been declared inside '::'
         friend void ::tud_hid_report_complete_cb(uint8_t instance, uint8_t const* report, uint8_t len);
                                                                                                      ^
In file included from c:\Users\wbadry\Documents\Arduino\libraries\esp32-usb-v2\src\hid\hid.cpp:1:
c:\users\wbadry\documents\arduino\libraries\esp32-usb-v2\src\usb_hid.hpp:86:102: error: 'void tud_hid_report_complete_cb(uint8_t, const uint8_t*, uint8_t)' should have been declared inside '::'
         friend void ::tud_hid_report_complete_cb(uint8_t instance, uint8_t const* report, uint8_t len);
                                                                                                      ^

Using library ESP32 USB V2 at version 2.0.0 in folder: C:\Users\wbadry\Documents\Arduino\libraries\esp32-usb-v2 
exit status 1

Compilation error: exit status 1
```

chegewara, 2024-03-06:
Hi
this version is quite a messy, because i wanted to make it compatible with arduino and esp-idf.
The thing is that arduino team decided to use different fork of tinyUSB with completely different set of flags in sdkconfig etc.
Because of that i decided to abandon this version, at least for now.

*offtopic*
Currently i am playing with ble mesh and trying to write user friendly C++ library. Again, it wont be arduino compatible, because ble mesh is disabled in there. There is still option to use ble mesh with nimble tho.

wbadry, 2024-03-07:
Thank you for your response. So there is no working version with SDIO interface? I am looking just for that, so the ESP32S3 could be recognized as USB mass storage device and be able to copy the files from that device.

chegewara, 2024-03-08:
In theory you could try to import sd card (sdio) code to old library. Only the code that initialize sdio peripheral.
That should works