chegewara/esp32-usb-v2 #12: hal/usb_hal.h missing in Arduino

imakin, 2024-09-09:
I tried for ESP32-S2 in linux and got this error. Any clue what is this usb_hal and how to include this?

        src/esp32-usb-v2/src/common/usb_device.cpp:3:10: fatal error: hal/usb_hal.h: No such file or directory
            3 | #include "hal/usb_hal.h"
              |          ^~~~~~~~~~~~~~~
        compilation terminated.
        
        Used library Version Path
        Wire         2.0.0   /home/user/.arduino15/packages/esp32/hardware/esp32/3.0.0/libraries/Wire
        
        Used platform Version Path
        esp32:esp32   3.0.0   /home/user/arduino15/packages/esp32/hardware/esp32/3.0.0
        Error during build: exit status 1
        make: *** [Makefile:23: compile] Error 1

i also tried the EspTinyUSB library and got the same usb_hal error like that.
#https://github.com/chegewara/EspTinyUSB/issues/157

chegewara, 2024-09-12:
Arduino 3/x is based on esp-idf 5.x, which changed components structure.
At the moment i do not have plans to support arduino v3.x for many reasons.

hwacookie, 2025-01-10:
Great, I just love how this magic works: As soon as you write down your question, you find the answer yourself.
Just in case anyone will need this (probably that will include future-me):

In your platformio.ini, use

[env]
platform = platformio/espressif32; Do NOT use https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF53
board = lolin_s2_mini

Can't remember why I tried the forked version.I surely had some very good reason.... :)

cabiamdos, 2025-05-01:
so the solution is to downgrade to the version esp32 2.0.17 board so that is compatible with this package isn't it?